codiac2600/SKR-MK3s-V1.4-Beta #2: SKR 1.3

dardan, 2020-02-18:
Hi Chris,
I used your firmware seems to work just fine with SKR1.3 tmc2130 and bltouch(Clone),
If you have already you bltouch configs its ok i will send my configurations tonight.
Keep the good things going

dardan, 2020-02-19:
[Configuration.h.txt](https://github.com/codiac2600/SKR-MK3s-V1.4-Beta/files/4228008/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/codiac2600/SKR-MK3s-V1.4-Beta/files/4228009/Configuration_adv.h.txt)

codiac2600, 2020-02-22:
Awesome! I haven’t tested the SKR V1.3 but I’m glad it’s working. I’ll run this on another machine when I get a chance.

n4mb3r0n3, 2020-02-24:
> [Configuration.h.txt](https://github.com/codiac2600/SKR-MK3s-V1.4-Beta/files/4228008/Configuration.h.txt)
> [Configuration_adv.h.txt](https://github.com/codiac2600/SKR-MK3s-V1.4-Beta/files/4228009/Configuration_adv.h.txt)
> 
> [Bltouch_Extruder.zip](https://github.com/codiac2600/SKR-MK3s-V1.4-Beta/files/4228077/Bltouch_Extruder.zip)
> [image]

Is this Part for the Original Prusa Extruder or something else ?

dardan, 2020-02-24:
@n4mb3r0n3  Yes this does fits with the original parts as the MK3s
[image] the difference is that is modified for the bltouch
[image]