codiac2600/SKR-MK3s-V1.4-Beta #5: i cant get p.i.n.d.a to triggers skr 1.4 turbo

angelshce22, 2020-02-22:
i have wiring like in the pdf and firmware uploaded but in proterface whit m119 cant get to trigger

codiac2600, 2020-02-22:
Is the light on the pinda turning on and off when a piece of metal is placed nearby?

angelshce22, 2020-02-22:
> 
> 
> Is the light on the pinda turning on and off when a piece of metal is placed nearby?

Yes

codiac2600, 2020-02-22:
I should have asked first. Did you compile your own firmware or are you using one of the final BIN firmware from this GitHub?

angelshce22, 2020-02-22:
> 
> 
> I should have asked first. Did you compile your own firmware or are you using one of the final BIN firmware from this GitHub?
own compile , to reverse steppers but dident changue any more

angelshce22, 2020-02-22:
@codiac2600 okay , whit Final BIN  works

closed 2020-02-22 by angelshce22