codiac2600/SKR-MK3s-V1.4-Beta #6: Extruder motor stops after first layer-ish

2BWorks, 2020-02-24:
Hey Chris,

Great Video, Great Firmware! I have followed it for my custom bear build and its working great. Now i try to do a print with your slicer profiles. But the extruder motor stops spinnen after the first layer while the X,Y,Z motors keep spinning.

I have tried other motors (happens to them aswell) tried other profiles (ender 3 profiles, and in these profiles the problems dont exist).

When hooked up to pronterface it does output the following:

```
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:Unknown command: "M73 P0 R30"
echo:busy: processing
echo:busy: processing
echo:Unknown command: "M73 P3 R29"
```
 
Do you think it has something to do with m73? (I think it just outputing progress to the display.)

codiac2600, 2020-02-24:
What drivers are you using? That seems like an issue with tmc2208 where you had to disable stealthchop. Also are you using marlin 2.0.3?

2BWorks, 2020-02-24:
They are running tmc2208's indeed but they are in standalone mode, so no uart

codiac2600, 2020-02-24:
Hmm. Basically for a long long time the tmc 2208 drivers have had an issue with stealthchop on the extruder when using it with linear advance. If you disable linear advance they should work but then you don’t get linear advance. Your options are to just swap the extruder stepper for something like the 2209, connect the 2208 via uart so you can disable stealthchop on the extruder only which still works great I run that on one printer, orrr swap all the drivers to tmc2209. All those should fix this.

2BWorks, 2020-02-24:
Hmm its weird though that using a different profile works fine. Without any firmware changes. I'm thinking the G90 has something to do with it. I'll try that first and afterwards ill try hooking it up to another step stick. I have some other drivers lying around. Got the TMC2209's on order but, well, because of the virus who knows how long they'll take to get here.

codiac2600, 2020-02-24:
That may be because the other firmware using using linear advance. Disable linear advance and it should work. So set K to 0.

2BWorks, 2020-02-24:
No its a different profile on the same printer :)

2BWorks, 2020-02-24:
On a total unrelated note; do you know how i can offset the X0 for like 20mm. Because im using couplers to attach the z rods to the z  motors. And my fan of the extruder assembly is bumping into the coupler.

I've tried adjusting the X MIN POS but im getting errors saying the print volume is to narrow.

2BWorks, 2020-02-24:
Edit: Sorry Chris, I have set the K value to 0 in the filament settings and it prints without a flaw now! Thanks for the suggestion

Closing issue!

closed 2020-02-24 by 2BWorks