codiac2600/SKR-MK3s-V1.4-Beta #14: MKS 1.4 Turbo / Homing X Y / MK2.5s

onkelfunny, 2020-03-14:
Hello Chris,

I have installed the MK3s version for MKS 1.4 Turbo. Currently the Bear 2.5s parts are installed. But I will replace them with the MK3s parts.

Z Homing works like it should.  With Y and X the axis moves only briefly. I have not connected the end stops. 

Here a link to a video: https://filebin.net/631rxdbz8si9yxiz/IMG_0119.MOV

What could be the reason?

Thanks

onkelfunny, 2020-03-14:
i have connected another motor without load. Unfortunately the same problem. It can't be the stepper driver, I replaced it several times.

Strangely enough, this problem only occurs when homing. The axes can be moved normally via the "Motion -> Move Axis" menu.

codiac2600, 2020-03-14:
Your setup runs 12V right now. You’ll need to make a custom firmware for 12V and then also tweak the homing currents and sensorless homing sensitivity.

onkelfunny, 2020-03-14:
I have compiled the firmware with your config files myself. I have already set CHOPPER_TIMING to CHOPPER_DEFAULT_12V. Do I have to change anything else for 12V?

I try to adjust the values for SENSORLESS_HOMING.

codiac2600, 2020-03-14:
Under the TMC settings increase the homing current.

onkelfunny, 2020-03-14:
I have tested these values

`
#define X_CURRENT       800
#define X_CURRENT_HOME  X_CURRENT
`

`
#define X_STALL_SENSITIVITY  10
...OR...
#define X_STALL_SENSITIVITY 200
`

Unfortunately still no success :(

codiac2600, 2020-03-14:
X current should be about 400mA. Try 300 as the homing current. 800 is waaaay too high.

onkelfunny, 2020-03-14:
no luck with this values :(

onkelfunny, 2020-03-14:
I have commented on SENSORLESS_HOMING. Unfortunately still no success. Which is also noticeable, if I turn the X axis home the motor turns very briefly in the wrong direction. In the direction of the power supply.

onkelfunny, 2020-03-14:
Hello Chris,
I found the problem. I cut the diagnostic pin from all the drivers. Didn't read your manual right.

Sorry for wasting your time. :(

closed 2020-03-14 by onkelfunny