codiac2600/SKR-MK3s-V1.4-Beta #15: Bed Leveling

onkelfunny, 2020-03-18:
Hello Chris,

can you tell me which version you are testing with the SKR board? For example, I can't get the bed level to work with version 2.0.3. If the 49 points are tested, the distances are all ok. When printing, I have extreme problems with the first layer. The distances to the bed always vary on the y axis and are sometimes 5mm.

Do I have to do something to calibrate the bed? I have activated the bed leveling under "Motion -> Bed Leveling" and also rebuilt the mesh.

I am sure that I have missed something... maybe you have an idea? Does your SKR printer print in general or do you have problems with it?

Thanks 
onkelfunny

codiac2600, 2020-03-18:
You need to set the Z offset. Probing gets the mesh values and you need to set how close the nozzle is to the bed. I have 6 printers running and they work pretty much spot on without issue.

onkelfunny, 2020-03-18:
the problem is that i have already set the z offset. the print usually starts very cleanly. but suddenly the nozzle moves to the wrong position.

just so i understand correctly. do you mean the function "PLA z offset" under "custom menu"?

Would be cool if we made a short video... "setup after firmware flash for dummies" :D

onkelfunny, 2020-03-18:
I have now completed: 

1st Full Calibration
2nd PLA Z Calibration => The result was this:

https://ibb.co/jwdQN1g
https://ibb.co/qyFLJDS

After that I started printing. The profile is from this repo, so I assume that it works.

The result looks like this:

https://ibb.co/svPj9J9
https://ibb.co/9y8vRwB
https://ibb.co/ZGgxG97

Now I have no idea what else it could be.

onkelfunny, 2020-03-24:
With the new Version 2.5.5.1 and UBL everything is working. over 40 prints without any issue :D
Thanks for the new config.