codiac2600/SKR-MK3s-V1.4-Beta #16: SKR MK3s wire guide

MarcoZ76, 2020-04-11:
Hi! If you want the hotend cooling fan switch on when it's over 50°C (or what you chose in the firmware) you should wiring the fan to the HE1 pin  (P2_04)  and change the guide like below:
[image]

reallost, 2020-04-11:
Thank you. In the video they discuss it but very briefly and not very clear for a newbi To marlin. It was easy to figure out once i realized i was using the always live plug and i could run my fan off the hot end port and have control. Shame you cant control the fan ports.

flow23, 2020-07-17:
Please remember, FAN0 and FAN1 are always on. E0 and E1 can be controlled via firmware.

ruedli, 2020-07-24:
These connectors switch PIN2 with a MOSFET to ground, if you want to use the Prusa (or Sunon) fans normally used in Bear, then you should not use the other pion (PIN1), as that is connected to VBB (=12V/24V). Instead I would use a pin with 5V or NPWR (the power from the 5V DCDC board used to power Neopixels).

Is that a good idea?
Is the 5V "polluted" by the fan, should I add a diode/capacitor?